mistyk/inavradar-ESP32 #2: Closer integration/cooperation with INAV

digitalentity, 2019-01-08:
We need to cooperate closer on this project. Looping also @DzikuVx, @fiam

KingKone, 2019-01-08:
Yes i think so too, nice to hear from you! 
I thought we are not ready yet to bother you ...

its all a bit messy right now ..

KingKone, 2019-01-08:
Camille and I have no real experience in doing something like this, we played around with mcus but not something at this scale

digitalentity, 2019-01-08:
@fiam is working on "flock" support for INAV, the only thing we need is hardware to exchange coordinates between UAVs (and a "standardized" way of doing it)

digitalentity, 2019-01-08:
Messy or not - let's get it working first. We can cleanup and refactor it later.

KingKone, 2019-01-08:
the lora modems are very nice, somebody in the goup also found very small ones without display .. but have you and info about flock, so i can learn what that is

mistyk, 2019-01-08:
Thanks to you, we keep in touch when first release will available , to merge with inav we have some functions to redraw because we are not expert on inav ;)

closed 2019-05-08 by KingKone