mistyk/inavradar-ESP32 #3: Other plane OSD indicator showing incorrect location

noagendafpv, 2019-02-14:
## Current Behavior
"Other plane" radar indicator (circle/dot) appears 180deg off axis. When flying towards the other plane, the dot is at the bottom of the screen, and when flying away it is in the top of the screen.

## Desired Behavior
The "Other plane" indicator show show in the top half of the OSD when ahead of the plane being flown and bottom half when behind.

## Suggested Solution

## Who does this impact? Who is this for?
Everyone flying version 0.1A3.

badandy1223, 2019-02-14:
Have the Arrow's behavior selectable in the CLI in 3 modes:
OFF- Arrow does not appear in the OSD
ROTATE - Arrow rotates to the direction of the nearest UAV (doesn't matter if 1 or 6 UAV's are flying, ALWAYS the closest.)
FIXED - Arrow does not rotate. Fixed pointing up on the screen.

putimir, 2019-02-20:
I'm baffled too :/ is it possible, that dot is you and the center arrow is the other plane?

KingKone, 2019-02-23:
180° issue seems fixed now

closed 2019-02-23 by KingKone