mistyk/inavradar-ESP32 #5: Speed indicator doesn't work

4stralay, 2019-02-15:
On the image show below... the speed indicator show 0 Km/hr. 
The iNav firmware  was given by Camile in Facebook messenger for Matek F405SE (Matek F405 Wing)

[image]

On the image below the speed indicator does work.
iNav firmware was INAV-Radar 0.1-alpha3 for Matek F405 (Matek F405 CTR)

[image]

mistyk, 2019-02-17:
Regression due to "Miles unity" bugfix.. removing Miles for the next to verify that

KingKone, 2019-02-23:
Seems to work in my last test.
Leave it open till Metric - Imperial is fixed

noagendafpv, 2019-02-25:
https://www.youtube.com/watch?v=6bWrJzgyxeY

Worked on my ground test unit, with speed from flying module updating. MatekF722SE Alpha5 on the plane, and MatekF405SE Alpha5.1 on the ground test unit. Both have 915mhz LoRa modules on Alpha5 FW.

closed 2019-04-26 by KingKone