mistyk/inavradar-ESP32 #6: iNav Radar OSD FW version displayed in Radar OSD

noagendafpv, 2019-02-25:
## Current Behavior
iNav FW version not displayed in OSD and makes it harder to troubleshoot.

## Desired Behavior
Display FW version in the corner(top right/left or bottom right).

## Suggested Solution
Add the FW target, example "MatekF722SE A5,"to allow for quicker verification of test video FW.

## Who does this impact? Who is this for?
This impacts all Alpha FW testers.

OlivierC-FR, 2019-05-09:
Thats a good idea I'll add this, all infos are available on the MSP, just have to pick them

KingKone, 2019-05-22:
as we get integrated into inav 2.2 that will be left our i think?

OlivierC-FR, 2019-05-22:
Since v1.3 the inav version shows at boot time !
Can close then, it's in.
Note that sometime iNav returns a weird number like 78.56.22, instead of the expected 2.2.0
It' likely on iNav side, and does not affect normal operations.

KingKone, 2019-05-22:
okii

closed 2019-05-22 by KingKone