mistyk/inavradar-ESP32 #8: 868Mhz module with crossfire ?

ZinhoRacing, 2019-03-29:
Hi,
I'm really interested in the project! nice job.
 
I'm just wondering if the lora modules on 868MHz can work with Crossfire that is using the same frequency.

Perhaps, it's beter to go to 433mHz modules to avoid any interference ?
If some one has an opinion, I'm in the process to buy 2 modules.

Thanks.

KingKone, 2019-03-29:
Hi, it's the same technology as crossfire. 

you should be able to used crossfire and our module on 868 MHz but for true longrange you maybe choose 433 MHz.

Also there is ravenLRS running on this hardware. If you like to have a look.

The most of our useres use 433 MHz I am (the developer of the esp firmware) using 868 MHz here in germany because is more common here.

If you have  more quesitions, just ask!

=D

KingKone, 2019-03-29:
Also the default antennas are not very good...

StewLG, 2019-03-29:
I'm guessing, but I would expect running two 868 / 915 lora modules running right next to each other to have problems. You would be listening for a relatively weak signal on the Crossfire, and transmitting a relatively strong signal with the Inav radar lora modem. This local strong signal might drown out the distant weak signal on the ground; this kind of asymmetry is likely to cause issues. Even more fun, I would expect these issues to show up only when in the air / at distance, which makes debugging really exciting.

Look at https://github.com/StewLG/CraftServices (a sister project to this one in many ways) to see my troubles getting two DragonLinks running next to each other for very similar (but not identical) reasons.

I'd suggest trying a 433 lora module instead of the 868 one, but I have no direct experience trying this. It could be that 433 and 868 will also have interference problems, since 433 * 2 = 866, which is very close to 868. 

These sorts of problems are why, ideally we'd have the Radio RX doing the intra-craft communications in addition to conventional ground control. Crossfire supports this, but in an expensive and proprietary way with TBS's FLARM support. I know RavenLRS intends to support it; does it yet?

Again, I'm just speculating. If someone has tried it, and it works, in the air and at range, great! I'm just doubtful until I see it working, having tried similar things and had them fail.

KingKone, 2019-03-30:
As i sad, for longrange use 433 MHz. 
But Crossfire uses frequency hopping and i think it avoids freqencies that are in use.
So it would be a nice test to check that with RTH set up properly :-P
One of the next version will have logging so it should not be that complicated to debug it. 

For your 2 dragon links, maybe you have to try to use different freqs like 867,5 and 868,5 as long as the law in your country allows it and you can set the modems like that. 

RavenLRS has something called FLOCK in the making, as soon as its public I may work on compatibility to our system.
FLOCK works more like the CraftService you pulled up - our system uses direct air-to-air communication, more like ADS-B in a real plane. All the hardware needed is onboard.

ZinhoRacing, 2019-04-01:
Thanks for all the informations.
I bought 868MHz modules as there was the only availables and they were cheap.
And as my radio link is the crossfire on 868MHz, I'll pay attention on testing and let you know the results.

closed 2019-04-01 by KingKone