mistyk/inavradar-ESP32 #9: Splash screen with Inav Radar logo

ZinhoRacing, 2019-04-26:
Hi,
While playing with the code, I have made a splash screen if you interested I can give it to you @KingKone.
It looks nice on the little display 😄 (and I agree it's not big deal and bit useless but cool 😄 )

[image]

KingKone, 2019-04-26:
wow nice, i like useless but cool things

ZinhoRacing, 2019-04-26:
Here is the code.

```
// Include
#include "inavradarlogo.h"

//  in void InitDisplay() procedure just after display.flipScreenVertically ();

 display.clear();
 display.drawXbm(0, 0, logo_width_s, logo_height_s, logo_bits_s);
 display.display();
 delay(2000);

```

That's all :smile:

[inavradarlogo.zip](https://github.com/mistyk/inavradar-ESP32/files/3122199/inavradarlogo.zip)

KingKone, 2019-04-26:
Thank you! What converter did u use for the picture?

ZinhoRacing, 2019-04-26:
I used a online converter tool : https://www.online-utility.org/image/convert/to/XBM
That's the only one that worked for me.

closed 2019-05-15 by ZinhoRacing