mistyk/inavradar-ESP32 #11: Reboot after a while release 1.01

ZinhoRacing, 2019-05-09:
Hi @KingKone,
I'm testing new release 1.01 and first of all, that's an impressive work :smiley:

I'm using the modules specified in the hardware section (Banggood 868MHz).

What I see is after some minutes the unit is rebooting. 
I'm using 4 units to test, all ok and then they reboot some times, not at the same moment but all of them do. I'm doing it on the bench, no FC connected to any unit. All units powered from my laptop (4 independant USB port, no hub).

KingKone, 2019-05-09:
Oh ok, all 4 modules the same behavior? 

Could you specify 'some minutes'? 
More like 10 or 5 minutes?

I will send that to Olivier.

OlivierC-FR, 2019-05-09:
I recall I had such reboots some time ago during tests, but that was on the stats page I think, the one that shows the timings for the nodes. Maybe a memory leak or corruption there. Do you remember what display page it was when it happened ?

KingKone, 2019-05-09:
could you keep a serial terminal open on you laptop, ans paste the output after the crash?

OlivierC-FR, 2019-05-09:
I added an uptime in seconds, and let it run for at least 20 minutes on pages 0, 1 and 2, and did not get any reboot. I compared with a timer on my phone to make sure it wasn't rebooting while I was away.
I'll keep an eye on this

[image]
[image]

ZinhoRacing, 2019-05-09:
Hello. I’ll try to log serial tonight.
I don’t remember if i was on the same page on all the units. I’ll pay more attention tonight.

ZinhoRacing, 2019-05-09:
I think i got one. 

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11584
entry 0x40078a60
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1445 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088553 on core 1

Backtrace: 0x40085e14:0x3ffbe820 0x40086041:0x3ffbe840 0x40088553:0x3ffbe860 0x400d6ad9:0x3ffbe8a0 0x400d487d:0x3ffbe8c0 0x400d1088:0x3ffbe8f0 0x400d10bc:0x3ffbe910 0x400d1711:0x3ffbe930 0x400d176a:0x3ffbe950 0x40080f25:0x3ffbe970 0x40081e25:0x3ffbe990 0x40089999:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11584
entry 0x40078a60
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1445 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088553 on core 1

Backtrace: 0x40085e14:0x3ffbe820 0x40086041:0x3ffbe840 0x40088553:0x3ffbe860 0x400d6ad9:0x3ffbe8a0 0x400d487d:0x3ffbe8c0 0x400d1088:0x3ffbe8f0 0x400d10bc:0x3ffbe910 0x400d1711:0x3ffbe930 0x400d176a:0x3ffbe950 0x40080f25:0x3ffbe970 0x40081e25:0x3ffbe990 0x4000bfed:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11584
entry 0x40078a60

ZinhoRacing, 2019-05-09:
Another one. After a reboot. I was playing with 4 units standard mode and 1 more in silent mode. 
[image]

OlivierC-FR, 2019-05-10:
This code is not what I run on my modules, I have nothing RTOS in there, neither CLI.
What is the release built from, the master branch ?

OlivierC-FR, 2019-05-10:
Did you get the firmware from the zip packages, or did you build it yourself from the master branch within the last 2 days ?

ZinhoRacing, 2019-05-10:
Hi,
I'm using the code from the zip package air-2-air-868-1.01.zip from last 2 days. Not buiding it.

KingKone, 2019-05-15:
try 1.2 hope its better, we have an installer now =D

ZinhoRacing, 2019-05-15:
1.21 tested Ok :smiley: no reboot seen so far. (868Mhz modules)
I like the Logo :smiley:

Would it be possible to go to host page when armed and only switch off display when alt is greater than threshold ( 10m for exemple). As it is great to see Vbat and mAh, Sat num when on the ground.

My setup below, cularis moto glider, I cut a window to see the display on the side (with old version of Inav Radar).

[image]

KingKone, 2019-05-16:
you have compiled it yourself?
we have some bug in 1.2 and 1.21 too i think, so please don't fly, 
FC will overoad.

ZinhoRacing, 2019-05-16:
Yes. I have atom and plateformio. Ok no problem I’ll only test on the bench. Thanks.

OlivierC-FR, 2019-05-16:
@ZinhoRacing I pushed the 1.30 few minutes ago, with some important tweaks, and also the iNav side appears to be more stable, at least on the bench. I launched the build for all iNav targets, will then put it on the dropbox same as usual.

ZinhoRacing, 2019-05-16:
Merci @OlivierC-FR I’ll have a go tonight.

KingKone, 2019-05-22:
seems to be fixed in new versions

closed 2019-05-22 by KingKone