mistyk/inavradar-ESP32 #12: Target for Matek F411-Wing FC

noagendafpv, 2019-05-22:
**Current Behavior**
F411-Wing FC only has 2 UARTs which are needed for UART1: Receiver and UART2: GPS. Leaving no UARTs for MSP comms to ESP32 module.

**Desired Behavior**
Find other pins that can be used for UART RX and TX  for MSP comm

**Suggested Solution**
Utilize S5, S6 or S7 pads since most F411-Wing FCs are used on small wings only needing S3 and S4 for elevon servos.

**Who does this impact? Who is this for?**
Flying wing pilots utilizing the F411-wing for it's small size, but still wanting to use the ESP32 modules for iNav Radar.

KingKone, 2019-05-22:
yes i have to look into that again

KingKone, 2019-05-22:
i have a F411 now, but no soldering iron for now

Dronek, 2019-05-23:
The target inav_2.2.0_MATEKF411_SFTSRL2.hex allows to use the LED pin as well as the ST1 pin as two softserial ports. However, they are both TX only, as far as I know. It would be more useful if those pins were used for one full duplex softserial port.

Dronek, 2019-05-25:
I build an inav 2.2.0 FW from the development branch where I assigned softserial TX to ST1 pad and softserial RX to LED pad. Not sure if it works as intended, though. The inavradar module gets correct data from inav but I don't know if inav gets the data from the module via the softserial RX. Will report, as soon as I have tested it in the field with other nodes.

If someone wants to test, you can find the file [here](https://drive.google.com/file/d/1H4ticT_OK8629iITF-KMJ-pgzvDvftTt/view?usp=sharing).
Make sure to enable softserial.

noagendafpv, 2019-05-26:
I would prefer to still have both SoftSerial ports available if possible, I use SmartAudio and S.port for telemetry.

IVData, 2019-11-04:
@Dronek have you tested in flight? I'm asking for #14

Dronek, 2019-11-05:
> @Dronek have you tested in flight? I'm asking for #14

It has been quite a while since my last flight but as far as I recall nav radar never worked properly with the F411.