mistyk/inavradar-ESP32 #14: Screen turns off periodically, says "Timing adjust"?

IVData, 2019-11-04:
Hi,

I have my radar attached to the full duplex serial port on a F411-wing, and my target plane has the same but on a normal UART.

When powering each plane up separately they work as expected. But when both are powered my screen turns off and on every few seconds and says something like "timing adjust" on the screen? The plane with radar connected to a UART does not do this.

We couldn't get the hud to display at all, not sure if the screen is a symptom?

KingKone, 2019-11-04:
F411 wing has only 2 hardware uarts, so i asume, something is connected to a soft serial? that would be a problem. The only way on the F411 is to use ppm for the transmitter, so you have a additional uart free for INAV-Radar
Which display is truning off, the OSD?

IVData, 2019-11-04:
Yes so we have:

* Plane 1: Sbus, gps on UARTs, radar on softserial. Radar OLED screen turns on and off with "TIMING ADJUST"
* Plane 2: PPM, gps, radar on UART2, sceen stays on.

Both radar units report correct gps sats, inav, and A/B.
I am using softserial because it was reported working in #12

KingKone, 2019-11-04:
F411 is not working correctly if softserial is used, sorry

IVData, 2019-12-10:
FWIW I've tested the build given in #12 and it works fine.

closed 2019-12-10 by IVData