mistyk/inavradar-ESP32 #17: Source code with 1.4 release is 1.3 source code.

IVData, 2020-06-28:
The source zip with ver 1.4 in the release section is in fact ver 1.3. Or at least the version string in it says 1.3, and it is missing `src/pixel.h` which was added to the dev branch before 1.4 was released.

KingKone, 2020-06-30:
Yes thats correct. The dev branch is the current version, the pixel.h contains all the bitmaps. It has nothing to to with the pixel OSD, If that was your question.

If you want to add FrSky pixel OSD, feel free. 

My own pixel OSD has never gotten off the ground. I have one part of the hardware ASR6501 but I never started coding. The CubCell module is more realtime capable then the ESP32. That would be needed to spit out the video frame on time. 

If you have any question, I'll be happy to help you.

IVData, 2020-07-12:
[image]
This is what I mean. Not sure if that was understood or not. If you're aware then thats fine.